dkoch83/Marlin-1.1.9_Anycubic_Chiron #1: pins.xxx.h

faschi, 2019-03-30:
Hallo!

Muss man die ganzen pins.xxxx.h die mit dabei sind, beim Chiron aber ja nicht benutzt werden alle mit auf das Board laden? Im Grunde müssten die pins.h, pins.ramps.h, trigorilla´s und noch 2-3 weitere reichen oder? Oder fehlt der Software irgendwas, wenn man diese files weglässt?

dkoch83, 2019-03-31:
@faschi 
Hallo, 
die pins.xxx.h sind HEADER Dateien. Diese enthalten Definitionen zu den Board Pins, gewählt werden sie über die [Configuration.h] 
```C++
// The following define selects which electronics board you have.
// Please choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_TRIGORILLA_CHIRON
#endif
```

Nicht benötigten pins.xxx.h Dateien haben keinen Einfluss.

Wenn man benötigte Dateien weglässt, wird beim Compilieren der Compiler Fehler ausgeben.

faschi, 2019-04-08:
Hi!
Danke für die Info! Ich bräuchte ggf. eine Hilfe beim Umbau meines Chirons auf dual - gäbe es die möglichkeit sie/dich direkt per mail zu kontaktieren? Das wäre fein, meine email ist mail [at ]cabriofans [punkt] com.
Danke und mfg
Robert

closed 2019-05-10 by dkoch83