dkoch83/Marlin-1.1.9_Anycubic_Chiron #3: SD Card Reader don't work.

cryoplay, 2019-05-16:
The SD Card Reader don't work.

Original Chiron with TMC2208 and Cable change.
What you need from me that i can help?

dkoch83, 2019-05-18:
Hallo, geht es auch nicht wenn du die SD Karte raus und wieder rein stecks und dann den Aktualisieren Button drückst.

cryoplay, 2019-05-19:
Ja. Keine Funktion,

Mehrere SD Karten probiert, auch ein kompletter Powercycle.

dkoch83, 2019-05-20:
wurde an der config was verändert?

cryoplay, 2019-05-20:
Nein.
Es wurde nichts angepasst und nichts verändert.

wayp55, 2019-06-17:
Ich habe das gleiche Problem. 
Ich benutze Octopritn wie du, es ist der einzige Grund, warum ich drucken sollte.

dkoch83, 2019-06-19:
Ich habe ein paar Änderungen vorgenommen, wegen dem SD Karten Problem. Bitte mal testen. 

Nicht vergessen config zur sichern.

wayp55, 2019-06-20:
Danke, an diesem Wochenende werde ich die tmc2208-Treiber auf die Version v1.2 aktualisieren (die ich bereits zu Hause habe) und ich werde die Firmware aktualisieren, um zu sehen, ob das Problem gelöst ist.
Danke

wayp55, 2019-06-23:
Greattttt :) the card reader works again. I installed tmc2208 v1.2 (before I had v1.0) 
I put v 1.36v on all controllers, but Y-axis sounds bad. 
What do you recommend?

dkoch83, 2019-06-23:
https://wiki.fysetc.com/TMC2208/
RMS Current (A) = 1.0
Reference Voltage (V): = 1.41

The Y Axis sounds a little bit, you can play with the acceleration to make it a little quieter

```
echo:Steps per unit:
echo:  M92 X79.93 Y100.08 Z396.10 E418.60
echo:Maximum feedrates (units/s):
echo:  M203 X100.00 Y100.00 Z20.00 E80.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X6000.00 Y6000.00 Z50.00 E30000.00
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P450.00 R3000.00 T450.00
echo:Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 Q20000.00 S0.00 T0.00 X10.00 Y10.00 Z1.00 E10.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
```

wayp55, 2019-06-30:
I have a "new" problem, when I active M900 K0.5 (linear avance), extruder stop work but sometimes. Some Pieces finish, anohter not. I put again M900 K0.0. I installed new drivers tmc2208 v1.2 to solve it, but continue the problem. Could you help me. Could be for ventilation or hot excessive? (bad fan working)

dkoch83, 2019-06-30:
TMC2208 Reference Voltage?
Have you the original bowden extruder?
Are the tmc2801 V1.2 from fysetc? 
have you set the tmc ref voltage at 5 Volt usb or at mains voltage?
at main voltage is the right way, but carefull.

add to Start G-Code at the bottom for test.
```
M201 X6000.00 Y6000.00 Z50.00 E30000.00 ;Maximum Acceleration (units/s2)
M203 X100.00 Y100.00 Z20.00 E80.00 ;Maximum feedrates (units/s):
M205  X10.00 Y10.00 Z0.50 E15.00 ;Advanced: Q<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
M204 P450 T450 ;set acceleration
```

wayp55, 2019-06-30:
TMC2208 Reference Voltage?  --> I put 1.36/1.38 Is very dificult put 1.41 exactly
Have you the original bowden extruder?  --> Yes is the anycubic original
Are the tmc2801 V1.2 from fysetc? -->  FYSETC 1th Store TMC2208 v1.2
have you set the tmc ref voltage at 5 Volt usb or at mains voltage? Jumper, if I remember ok, is main voltage, not in usb option

dkoch83, 2019-06-30:
@wayp55
1.5 A (Imax) / 1.41 = 1,0638 A (RMS)
https://wiki.fysetc.com/TMC2208/#calculator
1.06 A RMS = 1.49 Vref

**you kann ajust the ref voltage from 1.41 up to 1.49 V for all TMC2208 stepper drivers**

[Picture ANYCUBIC TriGorilla](https://duckduckgo.com/?q=tregorila&t=ffsb&iax=images&ia=images&iai=https%3A%2F%2Fdirectvoltage-e7b.kxcdn.com%2Fwp-content%2Fuploads%2F2018%2F04%2FANYCUBIC-TriGorilla-Main-board-Compatible-Mega2560-RAMPS1.4-4-3.jpg)

To adjust the vref exactly, is the Picture of the board. On top right (5A -), power off the chiron, unplug the power cable, unscrew (5A -) an put a little piece of kabel to the (5A -) kabel hole and screw tight. Wenn you have a alligator clip ore simulare vor your Voltmeter, clip it on the end of the small cabel. 
Then you kann plug the power sorce an power on the Chiron. Now you have one hand free vor ajust the ref voltage and messurment at the same time.

wayp55, 2019-07-05:
Hi, I'm having a new problem. 
One of the Z-axis motors rotates to go up, but not down. It started suddenly. I turned off the machine after a bad impression after activating the linear advance, and when I have turned it off again and try to print something the motor fails. Do you think it could be because of a defective driver?

dkoch83, 2019-07-05:
@wayp55 
swap Z (Z1) with E1 (Z2) to test because a defect driver

wayp55, 2019-07-07:
> 
> 
> @wayp55
> swap Z (Z1) with E1 (Z2) to test because a defect driver

Do it and work again. Thanks :)

closed 2019-07-20 by dkoch83