dkoch83/Marlin-1.1.9_Anycubic_Chiron #4: hello, one problem with your firmware on my Chiron: when i want to perform a auto bed level with the sensor, after homing the head moves too high and then the sensor stops before reaching the bed,

CMET64, 2019-08-28:


dkoch83, 2019-08-30:
@CMET64 
the 4 screws of your bed are tightened.
unscrew them a little, then the bed comes up.

gcopeland, 2019-09-22:
He's right. This breaks auto bed leveling. Completely ignores the babystepping from the advanced settings leveling menu.

Rickbar, 2019-09-22:
Hi Guys.  I had the same problem.  After replacing the board, stepper drivers and reflashing the firmware the problem persisted.  I was resorting to manually leveling, but then the problem started to happen during prints.  It turned out to be the wiring harness on the X-axis.  In short, the four sets of 4 wires is a terrible way to make a wiring harness that is continuously moving back and forth, the wires crimp and break inside the casings.  I made my harness with flexible silicone wires, new Dupont pins, and reused the connectors.  It is a big job, but I could not be any happier with the quality of my prints, and everything works fine now.  PS:...check the resistance of the wires associated with the leveling sensor in the X-axis wiring harness...from the hotend connector to the control box connector.

dkoch83, 2019-09-22:
In the file [Configuration.h]
#define Z_PROBE_LOW_POINT          -8 // Farthest distance below the trigger-point to go before stopping

This prevent that the hotend hitting the bed without sensor.

gcopeland, 2019-09-22:
Thank you for the follow up. I have determined I was suffering from a mild case of stultus cerebrum. The feature is in fact working as intended.

Been a while since I last leveled my bed. I simple failed to do it correctly.

Thank you for providing this repo. I'm using it slightly tweaked. I also didn't know about the low point setting. Which is good to know even though it turns out I won't need it.

closed 2019-11-27 by dkoch83