dkoch83/Marlin-1.1.9_Anycubic_Chiron #5: Filament Runout sensor

NellyGTI, 2020-02-15:
Thank you for the great adaptation!

I'm having an issue recovering from a filament sensor runout. The printer correctly moves up and to the rear of the print bed, then retracts the filament. However, the LCD appears to freeze up with the filament sensor warning message (doesn't respond to touch) and I'm unable to continue the print. I am printing via Octoprint and the Terminal window seems to indicates pressing a button to continue, but I'm not sure what it is referring to. Is there a command I can issue to continue?

cb3dmaker, 2020-02-16:
When my filament runout is set to true i have a continue message on display, i can print only whit false setting fw but the sensor of filament don.t work

NellyGTI, 2020-02-16:
Where do you set it to false in the configuration?

cb3dmaker, 2020-02-16:
line 968 configh

cb3dmaker, 2020-02-16:
my is not a solution after few minutes error filament come back

NellyGTI, 2020-02-17:
Ok and when you refer to line 968 in config, do you mean commenting out: 
#define FILAMENT_RUNOUT_SENSOR

this results in an error if I comment it out:
sketch/chiron_lcd.cpp: In function 'void FilamentScan()':
chiron_lcd.cpp:329:39: error: 'runout' was not declared in this scope
  if (last_printing_status == false && runout.filament_ran_out == true) {
                                       ^~~~~

dkoch83, 2020-03-08:
I have fixed the problem.

closed 2020-05-24 by dkoch83