dkoch83/Marlin-1.1.9_Anycubic_Chiron #6: Dual Z Homing issue

cryoplay, 2020-05-07:
Hello.
I have a Problem with the dual Z Homing.

I can move the Z Axis over the Display up. But when i want to home, only the left Motor moves Down. The right Motor doesnt move. When i change the cables. Same Problem on the other side.

When i manually help to home, then i can move both Motors up and down over the Display.
But a second try to home also doesnt work another Time.

dkoch83, 2020-05-24:
Hello.
Is the problem with the original firmware too or only in my firmware?
Have you select the right config file?

cryoplay, 2020-05-24:
I forgot to close the Issue. 
It was a defective Endstop that made some false signals.
So sometimes it worked sometimes not.
Not Software Problem......
Sorry for the inconvinience.

closed 2020-05-24 by cryoplay