dkoch83/Marlin-1.1.9_Anycubic_Chiron #7: Motor extruder stop

wayp55, 2020-05-09:
Hey, first of all, thanks for your work.
I put your firmware back on the anycubic chiron and when I started the first print from the SD the extrusion engine stopped. So I connected the octoprint and tried to print the same file (a calibration cube) and the extruder motor stopped at the same point.

If you want I can send you the .gcode file

Attached is a link to the photo of the place where it stops.
https://subefotos.com/ver/?bb36e8c37f833228c779f87ddca33146o.jpg

I'm waiting a few days in case he manages to solve the problem but I'll go back to the original firmware.
Thank you

dkoch83, 2020-05-24:
Hello und sorry vor the late writing.
I Think the problem is linear advance, with some TMC2208 drivers thee extruder stops.
I have had the same problem and change the extruder stepper with LV8729 modified with one pin cut to lower the micro steps.

vor testing you kann add (M900 K0) in gcode vor disable.

closed 2020-08-02 by dkoch83

wayp55, 2020-09-15:
Hi, I have purchased an LV8729 for the extruder, can you tell me any links to cut the pin or instructions to mount it correctly?
And what voltage should be put on the driver.
Thank you