dkoch83/Marlin-1.1.9_Anycubic_Chiron #10: Chiron with firmware 1.3.5, new LCD supported?

gaby64, 2020-10-23:
Is the new LCD of newer Chiron shipped with firmware 1.3.5 supported on this git?

dkoch83, 2020-11-13:
no

raindancer2204, 2021-01-05:
**Hi there,**

I have just compiled the dkoch93 github firmware and got it running on my Chiron.

I have the following display that should correspond to the new version right?

Attached is the picture for testing.

Regards Raini

[image]

illuminati05, 2021-03-17:
> **Hi there,**
> 
> I have just compiled the dkoch83 github firmware and got it running on my Chiron.
> 
> I have the following display that should correspond to the new version right?
> 
> Attached is the picture for testing.
> 
> Regards Raini
> 
> [image]

This display is the old version my friend. They changed some communication from the display to the board.

illuminati05, 2021-03-17:
[image]
[image]

illuminati05, 2021-03-17:
One way to find out is to check the communication strings in the serial monitor with editing the dkoch83 code

illuminati05, 2021-03-17:
http://user.support.anycubic.com/Attachs/Uploads/20210317/1615970032285954ea9293ae3e.zip

illuminati05, 2021-03-18:
Hey guys good news. I have fixed the problem with autoleveling with the new display (V0.0.2) dark edition. Auto levelling menu will not be shwon and executed on the display. I have disassembled the display and connted the serial connector from the display to a serial adapter. The only difference in the communication is that the strings send from the display has changed. Under chiron_lcd.cpp there are these string (A30, A23, ....). The new communication string for auto levelling has changed from "A30" (OLD DISPLAY) to "A36" (NEW DISPLAY). I just added the same code for "A30" at the end of the select case for "A36"

Just add this code after:

`case 34: // A34 bed grid write`

`//--------------------------------------------------------------------------
    case 36: // A36 auto leveling - NEW TOUCH DISPLAY (V0.0.2 - 1.3.5) #17.03.21
      #ifdef AUTO_BED_LEVELING_BILINEAR
      if( (planner.movesplanned()) || (card.sdprinting) ) {
        write_to_lcd_P(PSTR("J24\r\n"));  // forbid auto leveling
        } else {
        write_to_lcd_P(PSTR("J26\r\n"));  // start auto leveling
      }
      if(TFTcode_seen(TFTcmdbuffer, 'S') ) {
        enqueue_and_echo_commands_P(PSTR("G28\nG29"));
      }
      #else
      write_to_lcd_P(PSTR("J24\r\n"));  // forbid auto leveling
      #endif
    break;
    //--------------------------------------------------------------------------`

My printer now works withe the NEW display like the OLD display. I hope this helps :)

dkoch83, 2021-03-18:
Thank you, for fix the problem.
i have add the code to the repository

illuminati05, 2021-03-18:
Hey Thank you for your reply. It seems to be the only change for the display. I haven't found any other change. I will give you feedback if any other things occur. Best regardsDominik-- Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

illuminati05, 2021-03-18:
My excel sheet with differencesBest regards Dominik-- Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

closed 2021-05-24 by dkoch83