dkoch83/Marlin-1.1.9_Anycubic_Chiron #12: #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER causes errors

illuminati05, 2021-04-07:
Hey dkoch83. The Anycubic Chiron firmware is based on the 1.1.9 version. My anycubic chiron touch display is defect. A new one is not  available. In the configuration.h there is the option for #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER. When I uncomment this i get a lot of compilling errors. This would be fine if an alternative display can be integrated if the stock one (ugly thing) is defect. I have three 12864 display for replacement.

Thanks for your work
Errors

`
sketch\Marlin_main.cpp: In function 'bool ensure_safe_temperature(AdvancedPauseMode)':

Marlin_main.cpp:6964:7: error: 'lcd_advanced_pause_show_message' was not declared in this scope

       lcd_advanced_pause_show_message(ADVANCED_PAUSE_MESSAGE_WAIT_FOR_NOZZLES_TO_HEAT, mode);

       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

sketch\Marlin_main.cpp:6964:7: note: suggested alternative: 'advanced_pause_menu_response'

       lcd_advanced_pause_show_message(ADVANCED_PAUSE_MESSAGE_WAIT_FOR_NOZZLES_TO_HEAT, mode);

       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

       advanced_pause_menu_response

sketch\Marlin_main.cpp: In function 'bool load_filament(const float&, const float&, const float&, int8_t, bool, bool, AdvancedPauseMode)':

Marlin_main.cpp:7000:11: error: 'lcd_advanced_pause_show_message' was not declared in this scope

           lcd_advanced_pause_show_message(ADVANCED_PAUSE_MESSAGE_STATUS);

           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

sketch\Marlin_main.cpp:7000:11: note: suggested alternative: 'advanced_pause_menu_response'

           lcd_advanced_pause_show_message(ADVANCED_PAUSE_MESSAGE_STATUS);

           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

           advanced_pause_menu_response

Marlin_main.cpp:7009:11: error: 'lcd_advanced_pause_show_message' was not declared in this scope

           lcd_advanced_pause_show_message(ADVANCED_PAUSE_MESSAGE_INSERT, mode);

           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

sketch\Marlin_main.cpp:7009:11: note: suggested alternative: 'advanced_pause_menu_response'

           lcd_advanced_pause_show_message(ADVANCED_PAUSE_MESSAGE_INSERT, mode);

           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

           advanced_pause_menu_response

Marlin_main.cpp:7033:9: error: 'lcd_advanced_pause_show_message' was not declared in this scope

         lcd_advanced_pause_show_message(ADVANCED_PAUSE_MESSAGE_LOAD, mode);

         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

sketch\Marlin_main.cpp:7033:9: note: suggested alternative: 'advanced_pause_menu_response'

         lcd_advanced_pause_show_message(ADVANCED_PAUSE_MESSAGE_LOAD, mode);

         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

         advanced_pause_menu_response

Marlin_main.cpp:7072:15: error: 'lcd_advanced_pause_show_message' was not declared in this scope

               lcd_advanced_pause_show_message(ADVANCED_PAUSE_MESSAGE_PURGE, mode);

[... 197 lines cut ...]

Marlin_main.cpp:11142:7: error: 'lcd_advanced_pause_show_message' was not declared in this scope

       lcd_advanced_pause_show_message(ADVANCED_PAUSE_MESSAGE_INIT, ADVANCED_PAUSE_MODE_PAUSE_PRINT, target_extruder);

       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

sketch\Marlin_main.cpp:11142:7: note: suggested alternative: 'advanced_pause_menu_response'

       lcd_advanced_pause_show_message(ADVANCED_PAUSE_MESSAGE_INIT, ADVANCED_PAUSE_MODE_PAUSE_PRINT, target_extruder);

       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

       advanced_pause_menu_response

sketch\Marlin_main.cpp: In function 'void kill(const char*)':

Marlin_main.cpp:15213:5: error: 'kill_screen' was not declared in this scope

     kill_screen(lcd_msg);

     ^~~~~~~~~~~

exit status 1
'lcd_advanced_pause_show_message' was not declared in this scope`

illuminati05, 2021-04-09:
Ich bau gerade ne offizielle marlin 1.1.9 so um das die änderungen von dkoch83 drinnen sind, aber ohne chiron lcd. Dh man kann bei dieser variante ein normales 12864 anhängen. Pins muss ich noch prüfen. Kompilieren geht schon

illuminati05, 2021-04-09:
Und dh die anycubic adapter müssen weg...

illuminati05, 2021-04-16:
Thats a pitty that no one responses

dkoch83, 2021-04-21:
Sorry for the late answer,
I look not so often at the projetct.
I have rebuild my chiron with an BTT SKR 1.4 Turbo and  TFT35 V2.0 Touch Screen from BTT. 
At the moment i use the newest version of Marlin firmware. 

This firmware hear is only for the Chiron with the orginal display. 

The chiron Display controllse the chiron board via serial comunication an his owne commands and respons. therfor it was required to add some code to the orginal marlin firmware to handle the display.
And this code modifikation makes it imporsible to compile the firmware hear with other displays.  

You cann config a new version or a fresh version of the 1.1.9 of marlin firmware for the chiron board with your display but this cann take a lot of time.

dkoch83, 2021-04-21:
Entschuldigung für die späte Antwort,
Ich schaue nicht so oft auf das Projekt.
Ich habe mein Chiron mit einem BTT SKR 1.4 Turbo und einem TFT35 V2.0 Touchscreen von BTT umgebaut.
Im Moment benutze ich die neueste Version der Marlin Firmware.

Diese Firmware hier ist nur für den Chiron mit dem Original-Display.

Das Chiron-Display steuert das Chiron-Board über die serielle Kommunikation, seine eigenen Befehle und Antworten. Daher musste der ursprünglichen Marlin-Firmware Code hinzugefügt werden, um die Anzeige zu handhaben.
Und diese Code-Änderung macht es unmöglich, die Firmware mit anderen Displays zu kompilieren.

Sie können mit Ihrem Display eine neue Version oder eine neue Version der Marlin-Firmware 1.1.9 für das Chiron-Board konfigurieren, dies kann jedoch viel Zeit in Anspruch nehmen.

illuminati05, 2021-04-22:
Hey thanks for your answer. In the meantime i have taken an default 1.1.9 and copied the chiron code from you without the display and it compiles with the lcd 12864. So problem solved.

closed 2021-05-24 by dkoch83