dkoch83/Marlin-1.1.9_Anycubic_Chiron #13: Not a bug, but a question from a dev without a chiron ;)

knutwurst, 2021-05-16:
Hello @dkoch83 !
As I have seen, you did an excellent job of making the Mesh Bed Leveling work with the integrated functions of the Chiron LCD. You have my full respect for it, because it's not as easy as it looks.

I myself develop for a number of Anycubic printers based on Marlin 2.0.x. in a so called [unified Firmware](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S) and am trying to "blindly" implement ABL support for the Chiron. I really stuck to the Anycubic code and your implementation and tried to recreate the MBL. Unfortunately unsuccessful so far. The Menu works and the MBL starts, but then it crashes.

The worst part is that I don't own a Chiron myself. Debugging is almost impossible.

So I would be really happy if you could just take a look at my code. Maybe you will notice something right away that I missed?

I would also like to include you in the credits if that's ok with you. I'm also open to pull requests ;)

To get a first glimpse, please take a look [here](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/c183f9046621767c05a5bac2ca8f6413c037a18d/Marlin/src/lcd/anycubic_touchscreen.cpp#L1902).

Best,
Olli

(I know that you're german too, but since this is github and I don't know how to contact you, I made an issue)

dkoch83, 2021-05-17:
Hello,
test
Line 1936 --> HARDWARE_SERIAL_PROTOCOL_F((float)(Zvalue), 2);
Line 1964 --> HARDWARE_SERIAL_PROTOCOL_F((float)(probe.offset.z), 2);
Line 1972 --> HARDWARE_SERIAL_PROTOCOL_F((float)(SAVE_zprobe_zoffset), 2);

The Function ftostr32 have a fixed lenght 
0 looks 000.00 and 1.20 looks 001.20

I cant test it on my Chiron i have rplaced the mainboard with a 32 bit Board one.

gretings,
dkoch83

closed 2021-06-19 by dkoch83

WenKwen, 2021-06-21:
Hello, boss, personal compilation of marlin2.0. X is always wrong, my mage-s can't use 2225, tmc2225 default 32 subdivision, can you give me a 32 subdivision TMC dgus firmware?

WenKwen, 2021-06-21:
The nozzle stopped before it reached the X limit

knutwurst, 2021-06-21:
@WenKwen I answered to your issue here: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/issues/165

@dkoch83 I would like to thank you again for your help with the implementation of the Chriron features! I couldn't have done it without you! Many Thanks!.